rustsbi/rustsbi #10: Software Delegate IllegalInstruction Exception to Supervisor Mode

wyfcyx, 2021-02-09:
Now RustSBI just panic if the illegal instruction is not related to platform/privilege spec compatibility. It causes that when an illegal instruction occurs in application(U Mode), the kernel(S Mode) cannot catch and handle it since RustSBI does not delegate the exception to the kernel but just panic.

In my opinion, we can delegate(software) the exception to kernel  in RustSBI by setting `scause/stval` etc. and jumping to the address which `stvec` points to, although there are plenty of details to be considered...

luojia65, 2021-02-09:
Nice idea, will fix after new year

luojia65, 2021-02-11:
功能写好了，我顺便搓了个测试框架，过了年欢迎来测。新年快乐！

luojia65, 2021-12-21:
Feature implemented in separate project implementation (https://github.com/rustsbi/rustsbi-qemu/blob/0108fb70de3608d224f851951b7a2541d40571be/rustsbi-qemu/src/feature/transfer_trap.rs). Thanks for suggestion!

closed 2021-12-21 by luojia65