rustsbi/rustsbi #11: Handle Load/StoreMisalignment Exception in RustSBI manually on k210 platform

wyfcyx, 2021-02-13:
Due to the hardware limit of Kendryte k210 chip(it is okay on qemu), when we want to load a 64-bit value in to a general-purpose register(using ``ld`` instruction), we have to guarantee that the address is aligned to 8 bytes. However, it is somehow strict so that many librarys do not consider it, leading to some incompatibility(e.g. ``xmas-elf`` crate). The same issue may occurs in other instructions, such as ``lh``/``lw``/``sh``/``sw``/``sd``, but I have not try them yet.

Should we handle them in RustSBI by performing some aligned memory access manually and then merging the results?

luojia65, 2021-02-14:
It's not hard to implement in SBI. Could we research on ELF file reader crate to find out why it does not support unaligned access?

Gallium70, 2021-08-07:
我在基于RustSBI做的另一个平台（核心是Rocket）的启动器里面，实现了对非对齐访存的模拟，可供参考。适配RV64IC，有一些硬编码，没有支持浮点扩展和32位架构。
https://github.com/Gallium70/lrv-rust-bl/blob/master/src/trap.rs#L265-L365
https://github.com/Gallium70/lrv-rust-bl/blob/master/src/misaligned.rs

luojia65, 2021-08-08:
@Gallium70 非常好，可以作为非对齐访问模拟的一种实现。

luojia65, 2022-05-16:
Hello! As this issue is related to K210 platform, for further discussions we raise issue on rustsbi-k210 repository.

closed 2022-05-16 by luojia65