rustsbi/rustsbi #12: rustsbi qemu panic due to

imjustfly, 2021-02-23:
加载rustsbi 时panic，机器是一台m1 mac，不知道跟cpu架构有没有关系。

```
➜  os git:(main)  qemu-system-riscv64 \
                -machine virt \
                -nographic \
                -bios ../bootloader/rustsbi-qemu.bin
[rustsbi] RustSBI version 0.1.1
.______       __    __      _______.___________.  _______..______   __
|   _  \     |  |  |  |    /       |           | /       ||   _  \ |  |
|  |_)  |    |  |  |  |   |   (----`---|  |----`|   (----`|  |_)  ||  |
|      /     |  |  |  |    \   \       |  |      \   \    |   _  < |  |
|  |\  \----.|  `--'  |.----)   |      |  |  .----)   |   |  |_)  ||  |
| _| `._____| \______/ |_______/       |__|  |_______/    |______/ |__|

[rustsbi] Platform: QEMU (Version 0.1.0)
[rustsbi] misa: RV64ACDFIMSU
[rustsbi] mideleg: 0x222
[rustsbi] medeleg: 0xb1ab
[rustsbi-dtb] Hart count: cluster0 with 1 cores
[rustsbi] Kernel entry: 0x80200000
[rustsbi-panic] hart 0 panicked at 'invalid instruction, mepc: 0000000080003eba, instruction: 0000000030200073', platform\qemu\src\main.rs:458:17
[rustsbi-panic] system shutdown scheduled due to RustSBI panic
```

luojia65, 2021-02-24:
mret指令解析出问题了，这个问题很多人报道过，qemu5.2出现的问题，一个修复的方法是降级qemu到5.1以下，或者用4.x版本。编译最新rustsbi的0.2.0-alpha.1版不会解决问题。

imjustfly, 2021-02-25:
不了解这是qemu以后会修复的bug，还是设计如此，以后就这样了？

如果是后者rustsbi会尝试适配否？THX！

SKTT1Ryze, 2021-02-25:
RustSBI 是 RISC-V SBI 标准的实现，qemu 怎么做和 rustsbi 无关。  
RustSBI 只适配 RISC-V SBI 标准，不适配 qemu。

luojia65, 2021-02-25:
等qemu修复吧，要么降级，我这软件是按sbi标准设计的，不方便改

imjustfly, 2021-02-25:
OK, Thanks. Close issue.

closed 2021-02-25 by imjustfly

SKTT1Ryze, 2021-07-08:
最新版本的 rustsbi-qemu 已经貌似已经解决了这个问题：https://github.com/rustsbi/rustsbi-qemu