rustsbi/rustsbi #16: Legacy sbi_set_timer() behaves differently than new sbi_set_timer() on QEMU ?

D0ot, 2021-05-08:
Rustsbi version: at commit 269d1d32b4c0fd4900e6db1f070e7a2ce983dd14
QEMU version: 5.2.0

Minimalist test case:
```c
int main() {
  w_stvec((uint64_t)kvec_asm); // i put some printf in interrupt handler
  s_sstatus(SSTATUS_SIE);
  s_sie(SIE_SSIE | SIE_STIE | SIE_SEIE);
  sbi_legacy_set_timer(r_time() + 30000000);
  // struct sbiret r = sbi_set_timer(r_time() + 30000000);
  // printf("%l, %l\n", r.error, r.value);
  while(1);
  return 0;
 }
```

When i use the legacy sbi_legacy_set_timer(),  the code works as expected, outputs a lot.

When i use the new sbi_set_time(), it seems that timer interrupt was never triggered..

I have traced the code, the params were properly passed.

[image]
[image]

Is the difference an expected behavior? 

Thanks ~~~

luojia65, 2021-05-09:
Seems like a bug, will test and fix soon

Gallium70, 2021-07-06:
[Legacy set_timer](https://github.com/rustsbi/rustsbi/blob/master/src/ecall/legacy.rs#L33-L72) will config the `mie.MTIE` and `mip.STIP` accordingly, while the [set_time from Timer Extension](https://github.com/rustsbi/rustsbi/blob/master/src/ecall/timer.rs#L23-L45) won't.
Personally I believe the legacy set_timer implementation is correct.

luojia65, 2021-10-19:
How does this code behave on OpenSBI? @D0ot

luojia65, 2022-05-16:
Hello! As this issue is related to QEMU platform, for further discussions we raise issue on rustsbi-qemu repository.

closed 2022-05-16 by luojia65