rustsbi/rustsbi #22: More build guidance in README.md

tommythorn, 2021-12-20:
For someone who isn't using Rust for cross compilation on a daily basis, I found the README a bit lacking in how to build this.  I tried
```
$ rustup target add riscv64gc-unknown-linux-gnu
$ cargo build --target riscv64gc-unknown-linux-gnu
```
but that crashed out with:
```
    Updating crates.io index
   Compiling memchr v2.4.1
   Compiling regex-syntax v0.6.25
   Compiling lazy_static v1.4.0
   Compiling nb v1.0.0
   Compiling bare-metal v1.0.0
   Compiling bit_field v0.10.1
   Compiling void v1.0.2
   Compiling nb v0.1.3
   Compiling embedded-hal v0.2.6
   Compiling aho-corasick v0.7.18
   Compiling regex v1.5.4
   Compiling riscv-target v0.1.2
   Compiling riscv v0.7.0
error: failed to run custom build command for `riscv v0.7.0`

Caused by:
  process didn't exit successfully: `/home/tommy/projects/chipsalliance/workloads/rustsbi/target/debug/build/riscv-c23973cdc9c82cbb/build-script-build` (exit status: 101)
  --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }', /home/tommy/.cargo.x86_64-linux/registry/src/github.com-1ecc6299db9ec823/riscv-0.7.0/build.rs:22:10
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

luojia65, 2021-12-21:
Build with riscv64imac target would solve the problem. And yes I should mark them in readme file, as it would help to solve the problem. :)

tommythorn, 2021-12-30:
Thanks @luojia65.  Just FYI, I got
```
$ cargo build --target riscv64imac-unknown-none-elf
   Compiling memchr v2.4.1
   Compiling regex-syntax v0.6.25
...
   Compiling rustsbi v0.2.0-alpha.9 (/home/tommy/projects/chipsalliance/workloads/rustsbi)                              
error[E0658]: use of unstable library feature 'asm': inline assembly is not stable enough for use and is subject to change                
  --> src/hart_mask.rs:73:13
   |                                                                                                                    
73 |             core::arch::asm!("                                                                                     
   |             ^^^^^^^^^^^^^^^                                                                                        
   |                                                                                                                    
   = note: see issue #72016 <https://github.com/rust-lang/rust/issues/72016> for more information
   = help: add `#![feature(asm)]` to the crate attributes to enable     
...
```                                                                                                                        
and a lot more like this.  It seems I have to use nightly, but I failed to figure out how to do that without making nightly the default (via `rustup default nightly`).

luojia65, 2021-12-30:
@tommythorn You'd try to use a nightly newer than 2021-12-15. That nightly includes a stabilization of asm! macro which would land in future 1.59.0 stable (would be MSRV of cureent project).