rustsbi/rustsbi #31: 能否限定 RustSBI 的实现通过 &‘static dyn 传入模块的实现对象？

YdrMaster, 2022-06-01:
现在各个模块的 init 都使用一般的泛型，并且传入转移所有权的实现对象。然而这样的对象由于无法在库编译时确定尺寸，实际上无法存储，只能放在堆上，这要求 SBI 的实现必须提供堆，这对于某些嵌入式设备似乎不够友好。

既然目前 RustSBI 已经以全局的方式提供服务，基本上已经在事实上要求实现对象是静态的。能否直接要求实现传入 &'static dyn 来初始化各模块？RustSBI 本体不再负责保存这些对象，只是通过指针访问它们。

这样还有个好处是设置对象可以是原子操作更新指针，更快更安全。

YdrMaster, 2022-06-02:
#30 实现了一个 AMO 版本的 OncePtr，利用 AMO 的顺序保证指针 meta 先于值设置，后于值读取，并且修改了 HSM 的接口

luojia65, 2022-06-02:
感谢！我觉得既然每个接口都必须用&self去调用，并且Self是多上下文共享的，那么Send加Sync约束确实比较合理。有更好的想法吗？

YdrMaster, 2022-06-02:
> 感谢！我觉得既然每个接口都必须用&self去调用，并且Self是多上下文共享的，那么Send加Sync约束确实比较合理。有更好的想法吗？

修改了一些，为 `AmoOncePtr` 提供了更明确的定义，尤其是对于生命周期的约束。

YdrMaster, 2022-06-02:
已经在 [rustsbi-qemu](https://github.com/YdrMaster/rustsbi-qemu/commit/ff916db16808aa70aa66b68307aa05b5d12870c4) 测试了这个版本的 rustsbi，工作正常，但也没能解决什么问题（Barrier 依然如故）

YdrMaster, 2022-06-06:
原来，只要别认为 LegacyStdio 和别的模块有什么区别，LegacyStdio 就不会和别的模块有区别。不要强行将 Serial 指认成 Stdio，LegacyStdio 就不需要可变性。

luojia65, 2022-07-06:
我记得log这个库有比较好的&'static dyn做法，不知道能不能在我们的项目里参考。

luojia65, 2022-07-10:
已经在 https://github.com/rustsbi/rustsbi/pull/30 合并了，感谢参与讨论 :)

closed 2022-07-10 by luojia65