rustsbi/rustsbi #87: Build fails due to unstable features in Rust

bosswnx, 2025-02-06:
I just cloned the repository and followed the rustc version specified in the README, but I encountered this issue on the first compilation.

[image]

Environment:

- rustc version: rustc 1.65.0 (897e37553 2022-11-02)
- OS: macOS 15.0 (24A335)

It can be fixed by switching rustc to nightly and adding the following codes to `sbi-spec/src/lib.rs`:

```rust
// ... existing code ...
#![feature(is_sorted)]
#![feature(const_mut_refs)]
// ... existing code ...
```

I wonder if there is something wrong with src or README?

bosswnx, 2025-02-06:
Update: 

I switched my rustc version to the newest version `1.86.0-nightly (a9730c3b5 2025-02-05)`, then it can be compiled successfully.

It seems that README is out of date.

luojia65, 2025-02-07:
@bosswnx: You're right—the README.md descriptions are outdated. If you're willing, could you contribute to the RustSBI README by updating the MSRV (Minimum Supported Rust Version) value? This would be a big help for everyone! :)

bosswnx, 2025-02-07:
Ok, I will do it this night.

PR #92 (2025-02-09): docs: update MSRV in README.md
docs: update MSRV in README.md

Here are some unstable features used by RustSBI with the original MSRV `1.65.0` in README.md:

- `is_sorted`: stable at `1.82.0` (https://github.com/rust-lang/rust/pull/128279)
- `is_some_and`: stable at `1.70.0` (https://github.com/rust-lang/rust/pull/110019)
- `&mut` in const: stable at `1.83.0` (https://github.com/rust-lang/rust/pull/129195)

I tested by myself and found out that `1.83.0` and above can compile successful, while `1.82.0` cannot.

Fixes: #87.
closed 2025-02-09 by luojia65