rustsbi/rustsbi #105: 为RustSBI生态库添加C语言别名文档支持

luojia65, 2025-02-15:
在RustSBI生态库的开发实践中，开发者通常需要结合使用`sbi-rt`库中的函数与`sbi-spec`包中的常量来实现RISC-V SBI扩展功能。由于C语言缺乏原生的模块化命名空间设计，开发者普遍采用前缀命名法（如使用`sbi_cppc_read`而非简单的`read`或`cppc_read`）来避免不同扩展模块或外部库之间的命名冲突。而基于Rust语言特性的SBI生态包，则通过crate和mod的模块化设计实现了自然的命名隔离，因此其函数无需携带`sbi_`前缀，常量也无需`SBI_CPPC_`等前缀。

为帮助C语言开发者更顺畅地过渡到RustSBI开发环境，我们计划在库文档中增加对C语言命名的搜索支持。具体方案是为Rust函数和常量添加`#[doc(alias)]`元数据标签（例如`#[doc(alias = "sbi_cppc_read")]`），使C开发者能通过熟悉的命名习惯快速定位对应的Rust实现。实现示例可参考[此代码片段](https://github.com/neri/mysbi/blob/43e76bc02d1e8634423b698a9bf35b4aaa626637/src/lib.rs#L340)。

具体实施范围：

- [ ] 为`sbi-spec`包所有常量添加C语言命名别名标签
- [ ] 为`sbi-rt`库所有函数添加C语言命名别名标签

实现规范示例：

- 针对`sbi-spec`包：为常量`sbi_spec::base::GET_SBI_SPEC_VERSION`添加`#[doc(alias = "sbi_get_spec_version")]`
- 针对`sbi-rt`库：为函数`sbi_rt::get_spec_version`添加`#[doc(alias = "sbi_get_spec_version")]`

---

**Add C language alias documentation support for RustSBI ecosystem**

In RustSBI ecosystem development practices, engineers typically combine functions from the `sbi-rt` library with constants in the `sbi-spec` package to implement RISC-V SBI extensions. Given C language's lack of native modular namespace design, developers conventionally adopt prefix naming (e.g., using `sbi_cppc_read` instead of simple `read` or `cppc_read`) to prevent naming conflicts between different extension modules or external libraries. However, the Rust-based SBI ecosystem naturally achieves namespace isolation through crate and mod system design, thus its functions don't require `sbi_` prefixes nor constants need `SBI_CPPC_`-style prefixes.

To facilitate C developers' transition to RustSBI development, we propose adding C-style naming search support in library documentation. The solution involves annotating Rust functions and constants with `#[doc(alias)]` metadata tags (e.g., `#[doc(alias = "sbi_cppc_read")]`), enabling C developers to quickly locate corresponding Rust implementations using familiar naming conventions. Reference implementation can be found in [this code snippet](https://github.com/neri/mysbi/blob/43e76bc02d1e8634423b698a9bf35b4aaa626637/src/lib.rs#L340).

Implementation scope:

- Add `#[doc(alias)]` tags for all constants in `sbi-spec` with C-style names
- Add `#[doc(alias)]` tags for all functions in `sbi-rt` with C-style names

Implementation specification examples:

- For `sbi-spec`: Add `#[doc(alias = "sbi_get_spec_version")]` to constant `sbi_spec::base::GET_SBI_SPEC_VERSION`
- For `sbi-rt`: Add `#[doc(alias = "sbi_get_spec_version")]` to function `sbi_rt::get_spec_version`

monkey169, 2025-02-15:
let me do it，think you

Plucky923, 2025-02-20:
Hi, I'd like to take on this task. I'll submit a PR as soon as possible.

luojia65, 2025-02-22:
Implemented in https://github.com/rustsbi/rustsbi/pull/112 , closing this issue.

closed 2025-02-22 by luojia65