rustsbi/rustsbi #131: Update `sbi-testing` to use stable `#[unsafe(naked)]` syntax for naked functions

luojia65, 2025-04-22:
## 问题背景

Rust 编译器的最新稳定版本（基于上游 PR [rust-lang/rust#134213](https://github.com/rust-lang/rust/pull/134213)）已确定将 naked_functions 特性中的 #[naked] 属性标记为 unsafe，要求开发者使用 #[unsafe(naked)] 语法以确保代码安全性。目前，RustSBI 的 sbi-testing 模块仍在使用 Nightly 版本的 #[naked] 属性，导致在稳定版 Rust 工具链中构建失败（参见 [workflow 运行结果](https://github.com/rustsbi/rustsbi/actions/runs/14575137489/job/40879573527)）。

## 任务需求

- 代码修改：将 sbi-testing 模块中所有 #[naked] 替换为 #[unsafe(naked)]。
- 兼容性验证：确保修改后的代码在稳定版 Rust（最新版本）和 Nightly 版本中均能通过编译。
- 文档更新：若相关文档或注释涉及 #[naked] 的用法，需同步更新为 #[unsafe(naked)]。

## 贡献要求

- Commit Message 规范：遵循 [RustSBI 团队的提交规范](https://github.com/rustsbi/slides/blob/main/2025/reports/%E4%B8%BA%20RustSBI%20%E8%B4%A1%E7%8C%AE.md)（如语义化前缀、简洁描述等）。
- Pull Request 说明：
    - 引用本 Issue 和上游 PR 链接，说明修复的必要性。
    - 详述测试结果（本地及 CI 验证）。
- 代码质量：确保无 Clippy 警告，并通过现有测试用例。

## 参考链接

- Rust 上游 PR：https://github.com/rust-lang/rust/pull/134213
- 构建失败日志：https://github.com/rustsbi/rustsbi/actions/runs/14575137489/job/40879573527

---

## Context

The latest stable version of Rust (based on upstream PR [rust-lang/rust#134213](https://github.com/rust-lang/rust/pull/134213)) has marked the #[naked] attribute under the naked_functions feature as unsafe, requiring developers to use #[unsafe(naked)] syntax for safety guarantees. Currently, the sbi-testing module in RustSBI still uses the Nightly version of #[naked], causing build failures in the stable Rust toolchain (see [workflow logs](https://github.com/rustsbi/rustsbi/actions/runs/14575137489/job/40879573527)).

## Tasks

- Code Update: Replace all #[naked] with #[unsafe(naked)] in the sbi-testing module.
- Compatibility Check: Ensure the modified code compiles on both the latest stable Rust and Nightly versions.
- Documentation: Update any related documentation or comments referencing #[naked] to #[unsafe(naked)].

## Contribution Guidelines

- Commit Message: Follow [RustSBI team conventions](https://github.com/rustsbi/slides/blob/main/2025/reports/Contributing%20to%20RustSBI.md) (e.g., semantic prefixes, concise descriptions).
- Pull Request Description:
    - Reference this Issue and the upstream PR, explaining the necessity of the fix.
    - Include test results (local and CI verification).
- Code Quality: Ensure no Clippy warnings and pass existing test cases.

## References

- Rust Upstream PR: https://github.com/rust-lang/rust/pull/134213
- Build Failure Logs: https://github.com/rustsbi/rustsbi/actions/runs/14575137489/job/40879573527

Saramanda9988, 2025-04-22:
您好，我认为我可以完成这个任务，想要认领这个issue

luojia65, 2025-04-22:
@Saramanda9988 好的，后续有任何问题请通过GitHub与我们的维护团队联系。

Q-9923, 2025-04-22:
哈喽哈喽我也想试试呢

luojia65, 2025-04-22:
@Q-9923 RustSBI团队的其它项目也需要更新到最新的unsafe(naked)语法，欢迎贡献

Saramanda9988, 2025-04-22:
@luojia65 您好，我完成了#[unsafe(naked)]的修改，但在进行Clippy测试时在library\sbi-testing\src\lib.rs:5:12中报错error: the feature `naked_functions` has been stable since 1.88.0-nightly and no longer requires an attribute to enable.
我需要将#![feature(naked_functions)]这个多余的特性门删除吗，还是保留，它并不影响编译和测试，仅仅是无法通过Clippy测试

Saramanda9988, 2025-04-22:
@luojia65 您好，我完成了#[unsafe(naked)]的修改，但在进行Clippy测试时在library\sbi-testing\src\lib.rs:5:12中报错error: the feature naked_functions has been stable since 1.88.0-nightly and no longer requires an attribute to enable.
我需要将#![feature(naked_functions)]这个多余的特性门删除吗，还是保留，它并不影响编译和测试，仅仅是Clippy警告

luojia65, 2025-04-22:
@Saramanda9988 你好，需要删除多余的cfg feature选项，因为我们未来直接在stable下编译，不再需要nightly features

closed 2025-04-22 by luojia65